tdene/synth_opt_adders #69: Bad update?

macd, 2022-06-11:
Has there been a refactoring and an update but the a new README.md was not added? For instance, all of the png's are now missing and show as broken links in the readme.  Also the examples in the readme currently don't seem to work.

tdene, 2022-06-11:
Yes.

I was working on updated docs for v1.0, due to the huge amount of changes.<br>
I took the step to remove old documentation on Jun 6th, intending to update it the next day. It is now 4 days later.

The current ETA is to have re-hauled docs out by the end of this weekend, on Jun 13th.

Apologies for the interim status. The old docs had to be purged at some point, and I did it too early.

tdene, 2022-06-14:
Closed by #82

closed 2022-06-14 by tdene