tdene/synth_opt_adders #94: GraphViz attributes containing colon cause issues

tdene, 2022-06-19:
See pydot/pydot#258

It is strange that this issue occurs on some builds, but not others. Colab installs are fine, as is my local install.<br>
Yet this issue does occur for others.

tcal-x, 2022-07-15:
I hit this issue on my personal laptop, which is running Ubuntu 20.04.   I am using `venv`, and this is what I have installed:
```
$ pip3 list
Package       Version
------------- -------
graphviz      0.20   
networkx      2.8.4  
Pillow        9.2.0  
pip           20.0.2 
pkg-resources 0.0.0  
pptrees       1.0.10 
pydot         1.4.2  
pyparsing     3.0.9  
setuptools    44.0.0 
```

P.S. the CI tests for #95 just need your click to kick off...

closed 2022-07-25 by tdene

tdene, 2022-07-25:
My sincere apologies for letting this sit for so long.

Part of it is new job stress, which I have now overcome.<br>
Most of it is that I wanted to fix #99 in the same version number as this, but #99 proved difficult to debug.

I am now active once again!

PR #95 (2022-07-25): Minor changes in png() @ ExpressionTree and add_node() @ ExpressionGraph
Minor modification in add_node() @ ExpressionGraph to quote HDL attributes.
Added string type cast to prevent nx_pydot from terminating with error.
Closes #94.
commit 9b88ff4d (2022-07-25): Merge pull request #1 from raha96/fix

Minor modification in add_node. Closes #94.
commit f2e504b9 (2022-07-25): Minor modification in add_node() @ ExpressionGraph to quote HDL attributes. Closes #94.