tdene/synth_opt_adders #122: Is YosysAdder still supported?

tcal-x, 2022-08-19:


tcal-x, 2022-08-19:
A separate issue with YosysAdder:

```
from pptrees.YosysAdder import YosysAdder as forest

width = 9
f = forest(width, alias = "ripple")
f.gif("adder.gif")
```

output:
```
Traceback (most recent call last):
  File "x.py", line 7, in <module>
    f.gif("adder.gif")
  File "/media/tim/GIT/tdene/synth_opt_adders/venv/lib/python3.8/site-packages/pptrees-1.2.6-py3.8.egg/pptrees/ExpressionForest.py", line 532, in gif
    fout.write(self._repr_png_())
  File "/media/tim/GIT/tdene/synth_opt_adders/venv/lib/python3.8/site-packages/pptrees-1.2.6-py3.8.egg/pptrees/ExpressionForest.py", line 210, in _repr_png_
    return display_gif(self.trees)
  File "/media/tim/GIT/tdene/synth_opt_adders/venv/lib/python3.8/site-packages/pptrees-1.2.6-py3.8.egg/pptrees/util.py", line 411, in display_gif
    graph.png(*args, out=fname, **kwargs)
  File "/media/tim/GIT/tdene/synth_opt_adders/venv/lib/python3.8/site-packages/pptrees-1.2.6-py3.8.egg/pptrees/ExpressionTree.py", line 1011, in png
    pg = nx.drawing.nx_pydot.to_pydot(self)
  File "/media/tim/GIT/tdene/synth_opt_adders/venv/lib/python3.8/site-packages/networkx-2.8.4-py3.8.egg/networkx/drawing/nx_pydot.py", line 302, in to_pydot
    any(
  File "/media/tim/GIT/tdene/synth_opt_adders/venv/lib/python3.8/site-packages/networkx-2.8.4-py3.8.egg/networkx/drawing/nx_pydot.py", line 303, in <genexpr>
    (_check_colon_quotes(k) or _check_colon_quotes(v))
  File "/media/tim/GIT/tdene/synth_opt_adders/venv/lib/python3.8/site-packages/networkx-2.8.4-py3.8.egg/networkx/drawing/nx_pydot.py", line 199, in _check_colon_quotes
    return ":" in s and (s[0] != '"' or s[-1] != '"')
TypeError: argument of type 'int' is not iterable
```

tdene, 2022-08-24:
YosysAdder should be supported, but it looks like support for it has fallen by the wayside.

The failure of diagram generation is concerning. It's definitely related to #94, the question is just how.